jrullan/StateMachine #5: arduino ide compiler error

Lostkid0617, 2020-03-13:
Hi, there, I wrote a process control program using this library last year, it works fine by then. However, I try to make some modification to the program, and compiler error appears, I remove all modification, compiler error won't disappear. My target board is Arduino mega 2560, my Arduino ide version is 1.8.12, the compiler errors are as following:
Arduino: 1.8.12 (Windows Store 1.8.33.0) (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from C:\Users\17629\Documents\Arduino\libraries\StateMachine\src/StateMachine.h:1:0,

                 from D:\code\arduino\NEATOS\neat_rzcp_an\neat_rzcp\neat_rzcp.ino:28:

C:\Users\17629\Documents\Arduino\libraries\LinkedList/LinkedList.h: In instantiation of 'ListNode<T>* LinkedList<T>::getNode(int) [with T = Transition*]':

C:\Users\17629\Documents\Arduino\libraries\LinkedList/LinkedList.h:314:28:   required from 'T LinkedList<T>::get(int) [with T = Transition*]'

C:\Users\17629\Documents\Arduino\libraries\StateMachine\src/State.h:83:32:   required from here

C:\Users\17629\Documents\Arduino\libraries\LinkedList/LinkedList.h:160:9: error: cannot convert 'bool' to 'ListNode<Transition*>*' in return

  return false;

         ^~~~~

C:\Users\17629\Documents\Arduino\libraries\LinkedList/LinkedList.h: In instantiation of 'ListNode<T>* LinkedList<T>::getNode(int) [with T = State*]':

C:\Users\17629\Documents\Arduino\libraries\LinkedList/LinkedList.h:314:28:   required from 'T LinkedList<T>::get(int) [with T = State*]'

C:\Users\17629\Documents\Arduino\libraries\StateMachine\src/StateMachine.h:54:41:   required from here

C:\Users\17629\Documents\Arduino\libraries\LinkedList/LinkedList.h:160:9: error: cannot convert 'bool' to 'ListNode<State*>*' in return

exit status 1
Error compiling for board Arduino Mega or Mega 2560.

Does someone else have encountered this problem and how could I fix it?

paragchhajed, 2020-03-31:
Tried to run an example from the library. 

Got following error: Please help.

Arduino: 1.8.11 (Windows 10), Board: "Arduino Uno"

In file included from C:\Users\pachhaje\Documents\Arduino\libraries\StateMachine\src/StateMachine.h:2:0,

                 from C:\Users\pachhaje\Documents\Arduino\libraries\StateMachine\examples\arduino_state_machine\arduino_state_machine.ino:49:

C:\Users\pachhaje\Documents\Arduino\libraries\StateMachine\src/State.h:11:1: warning: 'typedef' was ignored in this declaration

 typedef struct Transition{

 ^~~~~~~

In file included from C:\Users\pachhaje\Documents\Arduino\libraries\StateMachine\src/StateMachine.h:1:0,

                 from C:\Users\pachhaje\Documents\Arduino\libraries\StateMachine\examples\arduino_state_machine\arduino_state_machine.ino:49:

C:\Users\pachhaje\Documents\Arduino\libraries\LinkedList/LinkedList.h: In instantiation of 'ListNode<T>* LinkedList<T>::getNode(int) [with T = Transition*]':

C:\Users\pachhaje\Documents\Arduino\libraries\LinkedList/LinkedList.h:314:28:   required from 'T LinkedList<T>::get(int) [with T = Transition*]'

C:\Users\pachhaje\Documents\Arduino\libraries\StateMachine\src/State.h:83:32:   required from here

C:\Users\pachhaje\Documents\Arduino\libraries\LinkedList/LinkedList.h:160:9: error: cannot convert 'bool' to 'ListNode<Transition*>*' in return

  return false;

         ^~~~~

C:\Users\pachhaje\Documents\Arduino\libraries\LinkedList/LinkedList.h: In instantiation of 'ListNode<T>* LinkedList<T>::getNode(int) [with T = State*]':

C:\Users\pachhaje\Documents\Arduino\libraries\LinkedList/LinkedList.h:314:28:   required from 'T LinkedList<T>::get(int) [with T = State*]'

C:\Users\pachhaje\Documents\Arduino\libraries\StateMachine\src/StateMachine.h:54:41:   required from here

C:\Users\pachhaje\Documents\Arduino\libraries\LinkedList/LinkedList.h:160:9: error: cannot convert 'bool' to 'ListNode<State*>*' in return

exit status 1
Error compiling for board Arduino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Lostkid0617, 2020-04-06:
> Tried to run an example from the library.
> Got following error: Please help.
> Arduino: 1.8.11 (Windows 10), Board: "Arduino Uno"
> In file included from C:\Users\pachhaje\Documents\Arduino\libraries\StateMachine\src/StateMachine.h:2:0,
>              from C:\Users\pachhaje\Documents\Arduino\libraries\StateMachine\examples\arduino_state_machine\arduino_state_machine.ino:49:
> 
> C:\Users\pachhaje\Documents\Arduino\libraries\StateMachine\src/State.h:11:1: warning: 'typedef' was ignored in this declaration
> typedef struct Transition{
> ^~~~~~~
> In file included from C:\Users\pachhaje\Documents\Arduino\libraries\StateMachine\src/StateMachine.h:1:0,
>              from C:\Users\pachhaje\Documents\Arduino\libraries\StateMachine\examples\arduino_state_machine\arduino_state_machine.ino:49:
> 
> C:\Users\pachhaje\Documents\Arduino\libraries\LinkedList/LinkedList.h: In instantiation of 'ListNode* LinkedList::getNode(int) [with T = Transition*]':
> C:\Users\pachhaje\Documents\Arduino\libraries\LinkedList/LinkedList.h:314:28:   required from 'T LinkedList::get(int) [with T = Transition*]'
> C:\Users\pachhaje\Documents\Arduino\libraries\StateMachine\src/State.h:83:32:   required from here
> C:\Users\pachhaje\Documents\Arduino\libraries\LinkedList/LinkedList.h:160:9: error: cannot convert 'bool' to 'ListNode<Transition*>*' in return
> return false;
>      ^~~~~
> 
> C:\Users\pachhaje\Documents\Arduino\libraries\LinkedList/LinkedList.h: In instantiation of 'ListNode* LinkedList::getNode(int) [with T = State*]':
> C:\Users\pachhaje\Documents\Arduino\libraries\LinkedList/LinkedList.h:314:28:   required from 'T LinkedList::get(int) [with T = State*]'
> C:\Users\pachhaje\Documents\Arduino\libraries\StateMachine\src/StateMachine.h:54:41:   required from here
> C:\Users\pachhaje\Documents\Arduino\libraries\LinkedList/LinkedList.h:160:9: error: cannot convert 'bool' to 'ListNode<State*>*' in return
> exit status 1
> Error compiling for board Arduino Uno.
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.

Lostkid0617, 2020-04-06:
I have tried run the example came within this lib, got the same result. please note this.

m1dst, 2020-07-21:
Please see https://github.com/ivanseidel/LinkedList/issues/33 for information on how you can fix this.  It is an issue with the LinkedList version you have installed.

jrullan, 2020-07-28:
Guys sorry that I missed this issue. @m1dst is right, at some point a later version of the LinkedList library introduced that error and apparently is has been corrected in its github. It's an easy fix. I'm closing this issue since it is not specific to the StateMachine library.
Thanks to @m1dst for pointing it out.

closed 2020-07-28 by jrullan