jrullan/StateMachine #6: transitionTo() not working?

MDRRC, 2020-03-21:
Hello

I use the transitionTo() like this

```
        else
        {
            stm.transitionTo(stPowerOn);
            XPNet.getLocoInfo(0, 3);
        }
```

but there is no state change... 

The state machine keeps in the state where it was as shown in the logging

> StateMachine::run() : 5
> StateGetLocInfo
> StateMachine::run() : 5
> StateMachine::transitionTo() : 3
> StateMachine::run() : 5

What I'm doing wrong?

MDRRC, 2020-03-28:
"Fixed" it.

 * Added boolean transition and this is set to true in the transition functions

In run()

 ```
 if (this->transition ==false)
  {
    this->executeOnce = (this->currentState == next)?false:true;
    this->currentState = next;
  }
  else
  {
    this->transition = false;
  }
```

The `int next ` caused the problem, it was set to state x, when state transition was called it was still x ...

HamzaHajeir, 2020-04-23:
Hi 

I had the same issue, It's now solved. Thanks for this.

But here I got another issue : 
Somewhere transitionTo break and stuck the code. I got no idea why..

MDRRC, 2020-04-27:
Hi

 * Forked the state machine
 * Updated the code 
 * Updated the arduino_state_machine example.

closed 2021-02-10 by jrullan