jrullan/StateMachine #11: Adding the ability to add this library to multi-file projects

Voha888, 2021-02-02:
When creating a project in VScode with the Platformio plugin, it becomes necessary to re-include the library header file in several files, for example, to use the extern directive.
The original version of the library does not allow this. I separated the class methods into .cpp files, now the library header files can be included in several executable files

Please, see my pull request #10

jrullan, 2021-02-10:
Thank you for your contribution! Merged.

closed 2021-02-10 by jrullan

ilianOptrel, 2025-01-31:
Release `1.0.13` incorporating this fix is not yet available on the PlatformIO registry. There, the current version is still `1.0.11`, so the library needs to be included via URL. Could you update to the newest release on PlatformIO registry?