jrullan/StateMachine #12: Arduino Compiling Issue with LinkedList

placebo3681, 2021-03-08:
Hi there. Using your library for state machines in an Arduino project and ran into an issue with the following error. I do have the latest LinkedList version available through the Arduino library. Unless perhaps I installed incorrectly? 

In file included from C:\Users\Daniel\Documents\Arduino\libraries\StateMachine\src\State.h:1:0,
                 from C:\Users\Daniel\Documents\Arduino\libraries\StateMachine\src\State.cpp:1:
C:\Users\Daniel\Documents\Arduino\libraries\LinkedList/LinkedList.h: In instantiation of 'ListNode<T>* LinkedList<T>::getNode(int) [with T = Transition*]':
C:\Users\Daniel\Documents\Arduino\libraries\LinkedList/LinkedList.h:314:28:   required from 'T LinkedList<T>::get(int) [with T = Transition*]'
C:\Users\Daniel\Documents\Arduino\libraries\StateMachine\src\State.cpp:46:32:   required from here
C:\Users\Daniel\Documents\Arduino\libraries\LinkedList/LinkedList.h:160:9: error: cannot convert 'bool' to 'ListNode<Transition*>*' in return
  return false;
         ^~~~~
In file included from C:\Users\Daniel\Documents\Arduino\libraries\StateMachine\src\StateMachine.h:1:0,
                 from C:\Users\Daniel\Documents\Arduino\libraries\StateMachine\src\StateMachine.cpp:1:
C:\Users\Daniel\Documents\Arduino\libraries\LinkedList/LinkedList.h: In instantiation of 'ListNode<T>* LinkedList<T>::getNode(int) [with T = State*]':
C:\Users\Daniel\Documents\Arduino\libraries\LinkedList/LinkedList.h:314:28:   required from 'T LinkedList<T>::get(int) [with T = State*]'
C:\Users\Daniel\Documents\Arduino\libraries\StateMachine\src\StateMachine.cpp:31:41:   required from here
C:\Users\Daniel\Documents\Arduino\libraries\LinkedList/LinkedList.h:160:9: error: cannot convert 'bool' to 'ListNode<State*>*' in return
  return false;
         ^~~~~
exit status 1
Error compiling for board Arduino Uno.

mfresia, 2021-03-18:
Hi, I confirm the Issue.

I'm using Arduino 1.8.13 on Linux

same error messages.

I resolved changing line 160 of LinkedList.h library version: 1.2.3

from: 	return false;
to:		return nullptr;

Now It compile.

OlivierBourdin, 2021-03-23:
Hi @mfresia 
Thank you very much for the fix. I had the same issue with Andrdoid 1.8.13 under Windows. I implemented your fix and now it works for me as well!

atoav, 2021-05-12:
I have the same issue also with 1.8.13 (but on Linux). Adding the suggested change to line 160 of LinkedList version 1.2.3 didn't work for me:

```
Using library StateMachine at version 1.0.13 in folder: /home/myusername/Arduino/libraries/StateMachine 
Using library LinkedList at version 1.2.3 in folder: /home/myusername/Arduino/libraries/LinkedList 
exit status 1
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
cannot convert 'bool' to 'ListNode<Transition*>*' in return
```

mccartes, 2021-07-01:
This fails to compile on arduino because the linked list library includes the following in a "tests.cpp" file:

#include <iostream>
std::cout<< "Tests pass"<< std::endl;

I am not that experienced but it does not appear that iostream is supported with Arduino.
when I comment out these two lines, then the whole thing compiles successfully