jrullan/StateMachine #13: Transition to any state

krzychoooo, 2021-04-05:
this code work ok
```
void loop() {
  machine.run();
  machine.transitionTo(S4);
  delay(STATE_DELAY);
}
```

but this code not work
```
void state0(){
  Serial.println("State 0");
  if(machine.executeOnce){
    Serial.println("Execute Once");
    digitalWrite(LED,!digitalRead(LED));
  }
  machine.transitionTo(S4);    //**problem**
}
```

bechtold, 2021-11-21:
I can confirm this.
```cpp
State* HIBERNATE = machine.addState([](){
  if (machine.executeOnce) {
    Serial.println("State: Hibernate");
  }

  // Go to Menu
  if (buttons.pressedA()) {
    Serial.print("Button A pressed ");
    Serial.print(machine.currentState);
    Serial.print(" ");
    Serial.print(machine.transitionTo(MENU));
    Serial.print(" ");
    Serial.println(machine.currentState);
  }

  leds.loop(0, 1);
});
```
Also tried with `machine.transitionTo(2);`

In both cases the output is  "Button A pressed 0 2 2" but it stays in State 0/HIBERNATE.

krzychoooo, 2021-11-24:
Without changing the code of this library, it can be done like this:
```
State* nextState = nullptr;
void loop() {
  if (nextState)
  {
    machine.transitionTo(nextState);
    nextState = nullptr;
  }
  machine.run(); 
}
...
...
void state0(){
  Serial.println("State 0");
  if(machine.executeOnce){
    Serial.println("Execute Once");
    digitalWrite(LED,!digitalRead(LED));
  }
  nextState = S4;
}
```