jrullan/StateMachine #14: Compilations fails on Arduino Mega

fkromer, 2021-05-12:
If I try to compile

```
#include <StateMachine.h>

void setup() {}

void loop() {}
```

for Arduino Mega with v1.0.13 I get `Compilation error: Error: 2 UNKNOWN: exit status 1`.

dbartelmus, 2021-06-18:
Reinstall Arduino (try with non Arduino IDE) and reinstall your board specific files

mccartes, 2021-07-01:
This fails to compile on arduino because the linked list library includes the following in a "tests.cpp" file:

#include <iostream> 
std::cout<< "Tests pass"<< std::endl;

I am not that experienced but it does not appear that iostream is supported with Arduino.  
when I comment out these two lines, then the whole thing compiles successfully

fkromer, 2021-07-02:
BTW: I don't need the lib anymore and discontinued investigating this issue.

r2d2hp, 2022-01-20:
I followed the step to allow StateMachine to compile and download to my Arduino but unfortunately is does not output anything to the serial monitor.  I have set the baud rate to 19200 in the program and on IDE but its make no difference.  I am a novice so on a steep learning curve so any help greatly received.  
[image]