jrullan/StateMachine #21: Using states function inside class

AhmedYousryM, 2024-03-08:
The code is working great, by using the function addresses. But if this function is a part of class, then the state machine could not handle this function. If it is needed to define a state machine to handle many objects at the same time. It is tedious to define many state machines for each working object.

I recommend to add a new feature to the code. So that the state machine can be a part of a class and uses the class methods.

sle118, 2024-03-24:
I've encountered the same challenge with the current state machine library's limitations. To address this, I've developed an enhancement that modifies the Transition struct and the State class to support std::function, enabling the use of bound member functions, lambdas, and free functions as callbacks. This update aims to significantly increase the library's flexibility and ease of use, especially for projects that benefit from encapsulating state logic and transition conditions within classes. I believe this solution could provide a valuable improvement to the library, offering a more versatile and user-friendly interface for managing state transitions and logic.

aaronfultonnz, 2024-11-13:
@sle118 I'm interested in this functionality for a project I'm working on. Is this modification something you can share?